bondus/KlipperToolboard #3: Can hat power back feed?

trevjonez, 2020-06-09:
Wanted to ask if you have considered adding a 24 to 5v regulator to the can hat possibly with two jumpers (one positive, one negative) which would feed 5v back into the pi to power it. thus eliminating the need for the micro usb or usb c cable to power the pi. Jumpers so it could be easily disabled or bypassed if that made sense for a given application.

Just thinking aloud as I am running pi 3b+ on my machines and utilize a small buck converter on the gpio pins (04,06) for power. Given the hat design photographed I would need to change some of my setup should I ever try to utilize a setup on par with what you have demonstrated here. but eliminating the extra dangling buck converter would be a nice to have.

bondus, 2020-06-10:
I have been thinking the exact same thoughts, the current CAN hat is a quick hack. 
There is a new versions of the hat in the works with (among other features) pins to solder an external buck converter. But I am still looking for a good, cheap, small, easily available 5V 3A converter that can do the job. The Pi is very picky on power delivery.

trevjonez, 2020-06-10:
I don't have hardware knowledge beyond the very basics but I have had a good experience with these cheap buck converters thus far. https://www.amazon.com/gp/product/B07RVG34WR

A point of friction with those is that you need a dmm you trust to adjust the voltage before attaching to the PI. [Somewhere on the internet it told me](https://magpi.raspberrypi.org/articles/power-supply) to be careful about voltage when powering via the GPIO pins since there is not a fuse or regulator through that path. I've no clue if that little buck is risking the 3b+'s life though i've yet to have an issue and generally trust my meanwell 24v supply's

what is the motivation for using an external part for power conversion vs adding it directly to the hat pcb?

closed 2020-06-11 by trevjonez