bondus/KlipperToolboard #5: Higher Voltage version

bung69, 2020-12-29:
Hi, 

Really like this project.

Do you think it would be feasible to support a wider range of voltages,  up to 48V or 60V  for use with large machines and motors?

raymondh2, 2020-12-30:
48 or 60 volt would require a different stepper driver. It would also likely require changes to other power circuitry.

bondus, 2020-12-30:
The stepper driver used, TMC2209, is limited to 29V. Higher voltage drivers require external mosfets which require more space. 
And the power components would have to be totally different as ray writes, below 35V you have a plethora of options. 
Higher voltages do require special care to avoid sparks and such that I know nothing about.

olinek2, 2021-01-18:
You can use https://www.trinamic.com/products/integrated-circuits/details/tmc261c-pa/
I am running it at 50 VDC without any problems :)

Mosher23, 2021-04-18:
TMC5160 can run up to 50V, so 48V would be really nice for CoreXY printers for X-Y Movements

theolas, 2022-03-11:
> TMC5160 can run up to 50V, so 48V would be really nice for CoreXY printers for X-Y Movements

As bondus said, those require external mosfets that take up more space.