bondus/KlipperToolboard #6: BL/Touch support

jstkatz, 2021-02-10:
Its a good concept, for my particular use being able to add a bl touch in addition to an endstop would be useful, the touch could remove an additional 2-4 wires though it does require being able to supply 300ma at 5V briefly

Along the same lines, support for an inductive probe might instead of the touch would likely be preferred by some, because they're unlikely to both be used you might be able to save some space by overlapping the header footprints or selecting the supply voltage pin as either 5 or Vin via a solder jumper

bondus, 2021-02-10:
When the board was designed klipper had no support for a probe or endstop on the MCU running the extruder, so endstops were very low priority. Now work is being done in klipper to allow that, it'll hopefully be finished soon.
For the next version I'll try to squeeze in 5V and certainly more and better endstop/probe support.

mraction, 2021-07-18:
In the current state of the V0.6, you can use BL-Touch, if you don't solder R19 und C2 and short R36. Then you can use Pin 3 from the J7 Connector for the Control-Pin from your BL-Touch.

Connect the sensor Pin from the BL Touch to Pin2 or Pin1 from the J7 Connector and how to connect 5V and GND should be clear. ;-)

Alternativ it should be possbile to use SWCLK or SWDIO from the debug port for the Control-Pin from the BL-Touch?

bondus, 2021-07-21:
Thanks for the hints. Looking at he BL-Touch docs it looks like all that is needed is a raw output PWM capable pin from the MCU, it will work with 3.3V for that PWM signal. 
I have considered moving the debug connector to pads on the back, they aren't really useful for normal use. And replace it with a generic GPIO connector with as many pins as will fit.
If it can support 2 normal endstops (X and Y on a corexy) and a bed probe, that'll cover most use cases. Bed probes are usually either an inductive (5V+), or a bl-touch, or some kind of switch.

arrowcircle, 2021-10-21:
> Thanks for the hints. Looking at he BL-Touch docs it looks like all that is needed is a raw output PWM capable pin from the MCU, it will work with 3.3V for that PWM signal. I have considered moving [...]

Hey @bondus! Thanks for the great work! Is bl touch already supported?

Seriyv0lk, 2021-12-08:
> Thanks for the hints. Looking at he BL-Touch docs it looks like all that is needed is a raw output PWM capable pin from the MCU, it will work with 3.3V for that PWM signal. I have considered moving [...]

I think you can use 5 volts as the main supply voltage on the board and get 3.3 volts to power the microcontroller and the driver with an LDO, it doesn't need so much current that you need a step down regulator.This solves the problem with powering BlTouch, limit switches and other equipment that needs 5 volts, and the microcontroller you are using is most pins tolerant to 5 volts.

iPeel, 2022-02-17:
I'm using a BLTouch with the 0.61 board, to drive the control pin I had to use the CLK ( PA14 ) pin and take GND, 5V and an endstop input from the endstop connector.

[bltouch]
sensor_pin: ^head0:PB10 # Endstop 1
control_pin: head0:PA14 # CLK