bondus/KlipperToolboard #13: Free up space on PCB

johnssarris, 2021-04-15:
You have so much crammed into this gem! If you need to free up space you might consider the following:

- Eliminate 2 of the mounting holes. Leaving 2 diagonal holes should be sufficient for mounting.

- Replace the MicroFit3 connector with one of the fancy NanoFit connectors.

brandonheller, 2021-04-15:
I'll give a contrasting opinion here - I had mounted a Huvud with the 4-pin pointing up, and realized recently that if I rotated it 90 degrees I can fit it much better into the available space.  Plus, I'm using all 4 holes to give rigidity to a mounting part.  With 2 holes, this wouldn't be possible.

bondus, 2021-05-01:
The NanoFit is a good alternative. It is a few mm narrower than the MicroFit. 
Regarding the holes I think it's best to keep them all. But I can make the pads a bit smaller and move components a tiny bit closer, as long as you stick to hex bolts the heads of the screws are not very big. Every little 0.5mm makes a big difference.

Grogyan, 2021-05-04:
PCBs can have parts on the bottom side too, which enables even more space

Pakue95, 2021-05-04:
The disadvantage of placing parts on both sides would be higher manufacturing costs and worse mounting options due to interference with the components on the bottom.

Grogyan, 2021-05-04:
True it doesn't leave much in the way of mounting options.
But also false.  

Slightly higher fabrication costs, but fairly negligible in volume manufacturing.

The alternative would be to have smaller parts.
Limitations on size are the connectors

I work in electronic manufacturing

bondus, 2021-10-06:
All good comment. It's designed for low volume <1k best case

closed 2021-10-06 by bondus