bondus/KlipperToolboard #16: Timeout on Wait for identify_response

antgeo, 2021-05-26:
Hi,

I'm having trouble establishing serial with the toolboard, the board is detected by ~/klipper/scripts/canbus_query.py but klipper fails to see a identify_response on start up. I've pasted my configurations below, it is interfacing with the Waveshare RS485 CAN HAT

> Created a socket
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
Timeout on connect

My environment:

> ========= Last MCU build config =========
CONFIG_LOW_LEVEL_OPTIONS=y
\# CONFIG_MACH_AVR is not set
\# CONFIG_MACH_ATSAM is not set
\# CONFIG_MACH_ATSAMD is not set
\# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
\# CONFIG_MACH_PRU is not set
\# CONFIG_MACH_LINUX is not set
\# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="stm32"
CONFIG_MCU="stm32f103xe"
CONFIG_CLOCK_FREQ=72000000
CONFIG_FLASH_START=0x8000800
CONFIG_FLASH_SIZE=0x10000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x5000
CONFIG_STACK_SIZE=512
CONFIG_STM32_SELECT=y
CONFIG_MACH_STM32F103=y
\# CONFIG_MACH_STM32F207 is not set
\# CONFIG_MACH_STM32F401 is not set
\# CONFIG_MACH_STM32F405 is not set
\# CONFIG_MACH_STM32F407 is not set
\# CONFIG_MACH_STM32F429 is not set
\# CONFIG_MACH_STM32F446 is not set
\# CONFIG_MACH_STM32F031 is not set
\# CONFIG_MACH_STM32F042 is not set
\# CONFIG_MACH_STM32F070 is not set
CONFIG_MACH_STM32F1=y
CONFIG_HAVE_STM32_USBFS=y
CONFIG_HAVE_STM32_CANBUS=y
\# CONFIG_STM32_FLASH_START_2000 is not set
\# CONFIG_STM32_FLASH_START_5000 is not set
\# CONFIG_STM32_FLASH_START_7000 is not set
\# CONFIG_STM32_FLASH_START_8800 is not set
\# CONFIG_STM32_FLASH_START_10000 is not set
CONFIG_STM32_FLASH_START_800=y
\# CONFIG_STM32_FLASH_START_0000 is not set
CONFIG_STM32_CLOCK_REF_8M=y
\# CONFIG_STM32_CLOCK_REF_12M is not set
\# CONFIG_STM32_CLOCK_REF_16M is not set
\# CONFIG_STM32_CLOCK_REF_25M is not set
\# CONFIG_STM32_CLOCK_REF_INTERNAL is not set
CONFIG_CLOCK_REF_FREQ=8000000
CONFIG_STM32F0_TRIM=16
CONFIG_CANSERIAL=y
\# CONFIG_STM32_USB_PA11_PA12 is not set
\# CONFIG_STM32_SERIAL_USART1 is not set
\# CONFIG_STM32_SERIAL_USART2 is not set
\# CONFIG_STM32_SERIAL_USART3 is not set
\# CONFIG_STM32_CANBUS_PA11_PA12 is not set
CONFIG_STM32_CANBUS_PB8_PB9=y
CONFIG_CANBUS_FREQUENCY=500000
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"
\# CONFIG_CUSTOM_STEP_DELAY is not set
CONFIG_INITIAL_PINS=""
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_INLINE_STEPPER_HACK=y
=======================

/boot/config.txt

>dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=2000000
dtoverlay=spi1-1cs

/etc/network/interfaces.d/can0

>auto can0
iface can0 can static
        bitrate 500000
        up ifconfig $IFACE txqueuelen 128

Thanks!

lkong, 2021-06-04:
Had the same thing happen to me, try stop klipper wait a few seconds and restart klipper. If that doesnt work, do a firmware restart.

antgeo, 2021-06-04:
Resolved this. the RaspberryPi did not have a clean power source.

closed 2021-06-04 by antgeo