bondus/KlipperToolboard #20: Adapt design to RPI pico (rpi2040 chip)

shaunmulligan, 2021-09-25:
Do you think it would be difficult or even feasible to adapt the design to make use of the rpi2040 chip rather than the stm32? It seems the stm32s are impossible to buy at the moment and one can pick up single units of the pi pico for about €1. I think the one potential issue is that the pico doesn't have a native CAN implementation but it seems there are multiple implementations of it using the PIO pins. 

Just wanted to get your thoughts on the possibility of the above, I am very interested in building a few of these boards but sourcing the parts is not super easy in the current environment :P

bondus, 2021-09-25:
The main issue with the rp2040 that I can see is the CAN bus, and the lack of 5V tolerant pins.
CAN bus should be doable with the help of the PIOs, but it's fairly complicated. I have not seen any full or even partial implementations of CAN bus on the rp2040 PIO. Interest is big so one will most likely appear, eventually.
Other than that it would be easy to CAD a new PCB and swap out the STM32.

shaunmulligan, 2021-09-28:
Yeah, I have had a look for a PIO based CAN implementation and it doesn't seem like it exists yet unfortunately but there do seem to be a number of people interested in it. Hopefully its just a matter of time 😄

closed 2021-09-28 by shaunmulligan