bondus/KlipperToolboard #21: JLCPCB BOM/POS

necrogami, 2021-10-06:
MWSA0503-100MT according to LCSC is discontinued so not available anymore.
There are 2 SMD Resistors (R4,R7) that don't even show up as parts.

L4 and U3 show as no inventory available. Any possibilities of changing the pcb so that other parts can replace them?

bondus, 2021-10-06:
Things change a lot on component availability.

The resistors should be easy to find replacements for. R4 and  R7 are the current sense resistors for the stepper, quite critical components, you can use different resistances for them. 1% and 1/4W if I remember right.
Check the TMC datasheet for what's required and klipper documentation for how to configure.

U3, TPS54202/TPS54302/TPS54308 is out of stock everywhere I'm afraid.  L4 is a coil related to that.

I have been working on fitting a TPS54331 instead of the TPS54302, it does fit. But as a CS engineer it will take me 3 prototype rounds to get it right. It's tricky to get those voltage regulators to behave. Know any EEs?

necrogami, 2021-10-08:
I don't know any EE's i do know a CE, i studied electronics eons ago in hs but i figure i've lost more then remembered. I can ask around.

closed 2023-08-07 by necrogami