bondus/KlipperToolboard #22: Steathburner

Dragi2k, 2022-01-17:
Heyho,

is it possible to hook up a chamber thermistor on the board ?
and is it possible to control led via the board ? (5v excluding)

and wen it is possible how ?

Thank you

iPeel, 2022-02-22:
Yes, there's a second thermister input on PA1 if you wanted a roaming chamber temperature thermister, although IMO one on the gantry would suffice?

For controlling the Neopixel, there's GPIO pins you can steal from the diagnostics header, SWDCLK ( PA14 ) and SWDIO ( PA13 ) could both be used for this and the 5V output on the endstop header is rated for 2A, the Neopixels on the SB BOM would only use 210ma.

Unsure about the tacho signal, maybe one of the endstop pins would be OK for this?

Dragi2k, 2022-02-26:
thx for the anwere but i have the 0.50 not the 0.6 version ;-)