bondus/KlipperToolboard #23: Documentation updates for V0.61 version of board

eddietheengineer, 2022-02-04:
Hello! I love the Huvud boards--I have a ton of the V0.50 on my printers and just installed the first V0.61 on a printer last night! There were a few gotchas/clarifications as more people install this:

Add flashing instructions
-8Mhz clock for make menuconfig build
-Add note -- if the wrong bootloader option is selected and you try to flash the Huvud, it will not work--need to flash the bootloader again using an ST-Link

Updated pin descriptions:
Thermistor 1 - PA0
Thermistor 2 - PA1
Heater - PA6
Fan 1 - PA7
Fan 0 - PA8
Endstop 1 - PB10
Endstop 2 - PB11
Endstop 3 - PB12 
ADXL CS pin - PB1
Diag for TMC2209? - PA15

rlaferla, 2022-03-03:
There are three pins on the thermistor, what is the mapping?
[image]

rlaferla, 2022-03-07:
[image]

rlaferla, 2022-03-18:
I can't seem to flash it.  

[image]

[image]

rlaferla, 2022-03-18:
Newer Waveshare boards have a 12M clock and not a 8M!

rlaferla, 2022-03-19:
I was able to get it to work.  The crystal speed in menuconfig must be 8M because it's the Huvud crystal and NOT the Waveshare one (which was 12M in my case.)

rlaferla, 2022-04-05:
SWDIO PA13?

charlespick, 2022-05-22:
+1 would like the docs to be updated with more detailed information about how to use the new adxl, single canbus/power connector, multiple endstops and thermistors, and terminating the canbus using the pads on the back.

dpoulson, 2022-11-05:
> I was able to get it to work. The crystal speed in menuconfig must be 8M because it's the Huvud crystal and NOT the Waveshare one (which was 12M in my case.)

Could you say how you managed to get it to work? I'm having the same issues, using the same settings as you. Output from make flash is exactly the same as above. (with different size of firmware of course)