bondus/KlipperToolboard #26: Inductive probe?

rlaferla, 2022-02-16:
Does this board support an inductive probe (Omron)?

rlaferla, 2022-02-21:
I was hoping to get some info here.  I see there is an endstop connector.  Can I use it for a Klicky probe?

iPeel, 2022-03-17:
You should be able to use an "n type" inductive probe, but Klicky or Euclid is an easier fit. On the 0.61 board there are three endstops and any one of them will work for the probe. I've use BLTouch and Klicky but haven't tested an inductive probe.

cjkzwe, 2022-07-23:
@iPeel Do you know how to write config if I want to use Klicky as a probe and connect two endstops at the same time on the 0.61 ver board?

iPeel, 2022-07-25:
@cjkzwe that shouldn't be a problem, there's three endstop inputs so you can use endstop 1 (PB10) for Klicky, endstop 2 (PB11) for X and endstop 3 (PB12) for Y.

example snippets:

#Huvud MCU definition
[mcu head0]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_52FF6D064850775026211367-if00
restart_method: command

[probe]
pin: ^head0:PB10

[stepper_x]
endstop_pin: head0:PB11

[stepper_y]
endstop_pin: head0:PB12

Not sure if you'll need ^'s on the endstop pins as I haven't converted mine yet.

cjkzwe, 2022-07-26:
@iPeel Thanks a million! I'll try this config as my board is ready.