bondus/KlipperToolboard #28: USB details

ModMike, 2022-02-18:
Can you provide some documentation of the USB functionality? I only need 1 tool head board so I don't need to daisy chain it.

1) How do I enable it?
2) Do I need to use USB power lines in addition to the main power?

Thanks for an AMAZING board. Best idea ever.

iPeel, 2022-02-20:
If you look at the back of the 0.61 PCB there's 6 pads in two columns between the Microfit3 connector and the USB connector but on the other side of the PCB. The top 2 pads are the USB D+ and D- and the bottom pads are the Canbus + and -, the middle are the Microfit3 data pins. There are tracks between the canbus and microfit pads, cut these with a knife then solder bridge between the USB and the Microfit pads, this effectively turns the microfit3 connector into running USB instead of Canbus.

You can then wire the microfit3 cable up to a USB type A, but don't take the ground from the USB or the power from the USB, just the D+ and D-. I used some sheilded cable to run from the USB, complete with the sheild but didn't terminate the sheild at the Huvud end. USB D+ is on the same side as the 24v+ and USB D- is on the same side as the power -.

Then, follow the instructions for flashing the Huvud board ( STM32, 2K bootloader etc ) but instead of using Canbus, choose to use USB. You can even flash the Huvud board over the USB going to the Microfit3 connector and this will confirm the USB is working.

To configure Klipper it's the same as any other USB board, look for its identity in /dev/serial/by-id and set this as the Serial in the MCU configuration:

[mcu head0]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_53FF6D064848885546352367-if00
restart_method = command

ModMike, 2022-02-20:
Thank you SO MUCH! I am fine with the CAN hat (have 3 on the way) I just wanted to avoid using GPIO in case I need them in the future. I understand that I am only using 2 and I can add a ton of devices on CAN. It also seems like USB will be easier to maintain via flash.

clawszilla, 2023-05-04:
Should I still have continuity between the d+ and d- and the Vin and GND on the microfit connector pins? Or is that going to cause a problem when I wire it to 24v? I'm worried I cut the pads too deep or something.