bondus/KlipperToolboard #33: Questions: Alternate TMC driver setup

flackenstein, 2022-02-27:
What are you thoughts on using a TMC driver module with a low profile header?  This way if the driver goes bad it can be quickly and easily replaced.  This also allows us to upgrade the driver module to alternate versions.

Example Module: https://www.digikey.com/en/products/detail/trinamic-motion-control-gmbh/TMC2209SILENTSTEPSTICK/10232488?utm_adgroup=Essen%20Deinki&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_DK%2BSupplier_Other&utm_term=&utm_content=Essen%20Deinki

Example Low Profile Header: https://www.digikey.com/en/products/detail/samtec-inc/TLW-108-06-G-S/1105092

Other benefit, if you use taller headers for the TMC driver you can place component underneath to maximize footprint.

flackenstein, 2022-02-28:
Worth adding it's a lot easier to procure the Trinamic SilentStepStick in the Americas and it has a unit cost that isn't that much more than the IC itself, and makes for easier repairs/upgrade.

https://www.trinamic.com/support/eval-kits/details/silentstepstick/

bondus, 2022-02-28:
There are a few downsides to having a stepstick/modular solution: 
1. Thermals is a lot worse. When mounting the driver directly to the board the whole board acts as a heatsink
2. Cost and complexity of a modular approach is higher.
3. Connectors use up a lot of precious board space. As you say some can be mitigated by the ability to put components under the module.

For now I think the better solution is to have the driver directly mounted to the main board.