bondus/KlipperToolboard #35: Thermister accuracy at low temperatures ( E3D V6 )

iPeel, 2022-02-28:
I have an issue with the thermister reading at low temperatures ( below 10 degrees celcius ). When I start up Klipper shuts down through an ADC threshold error, I have to blast the hot end with a heat gun to get the firmware to start. It seems the thermister on my 0.61 board is out by about 10 degrees at lower temperatures compared to the bed temps but fine at higher temperatures.

The pertinent bits from my extruder section are:

heater_pin: head0:PA6
sensor_type: ATC Semitec 104GT-2
sensor_pin: head0:PA0
pullup_resistor: 2200
min_temp: 0
max_temp: 300

Anyone else find this, or am I configuring something wrong?

iPeel, 2022-03-17:
In the end I resolved this by lowring the pullup_resistor value by a couple of hundred ohms in the Klipper config. Seems to report fine at both low and high temps.

jakep82, 2022-03-23:
I'm having a similar issue, and I've done some testing to confirm something isn't right.  I connected a PT1000 and a ATC Semitec 104GT-2  to TH1 and TH2 on the Huvud, and installed them in the same heater block for testing (both configured with 2200 ohm pullup).  In a 20C room, the PT1000 shows around 20-21C, but the thermistor is reading around 13-14C.  Using the PT1000 to control the hotend and heating it up so the PT1000 reads 230, the thermistor reads around 250.  I've checked resistance values on both the thermistor and the PT1000 at a couple known temperatures (0C and 20C), and both seem within tolerance of expected values.   

The PT1000 is definitely reading too low at operating temp, because printing ABS at my normal 235C temp resulted in print issues I associate with overheating the filament.  By contrast the thermistor is reading way too low at room temp, but seems like it might be accurate at normal print temps.

I'll be receiving a thermocouple today to do more testing.

iPeel, 2022-04-12:
@jakep82 that's interesting, let me know what your testing comes up with. Did you try amending the pullup_resistor value in ther config and if so what effect did that have on the top end readings?

jakep82, 2022-04-12:
I tested a Dragonfly and a V6 with my thermocouple inserted down into the nozzle of each hotend.  Both thermistors I tested read 5C low at 235C (Klipper reporting 235C on thermistor, thermocouple reading ~240C).  Most of this can likely be attributed to the fact that nozzle is in between the heater and the thermistor.  

My PT1000 by contrast was reading 10C low at print temps (235C on PT1000, ~245C on thermocouple).  This is why my prints seemed too hot with the PT1000.

In the end I'm using the stock 2200 resistor value with regular thermistors.  Somewhere around 50-60C the values on the thermistors correspond directly with the thermocouple.  This is in the typical range for my chamber temps, so it all works out for me.  I just ignore the room temperature readings.

iPeel, 2022-05-03:
@jakep82 thanks, this inspired me to buy a decent thermocouple and measure mine too from the inside of the nozzle. You are correct in that with the stock 2200 pullup config the values are out at the lower end but at operating temperatures seem accurate with within a few degrees. With an amended pullup resistor config the values were accurate at the low end on startup but were off by 10-15 degrees at operating temperatures.

I've reverted to the default config now, but it means in the winter when I start up in a cold workshop I'll have to hit the hotend with a hot air gun before switching the printer on to avoid startup errors.

jakep82, 2022-05-03:
@iPeel Just change the min_temp on all your sensors.  No need to use a heat gun.  

I should also mention I ended defining a custom thermistor to improve accuracy.  I used the QUERY_ADC function in Klipper to pull the raw values with a 2200 pullup at known temperatures I was measuring with my thermocouple.  By defining values at room temp, 50C, and 240C my thermistors are now accurate to within a few degrees throughout the range where I use them, including room temp.

iPeel, 2022-05-05:
@jakep82 I've never succeeded in configuring a min_temp below 0 degrees, I don't think it's allowed. In the winter my workshop drops to only a few degrees and this results in a negative reading on the Huvud. At the moment it's not an issue but later in the year I'll probably resort to a custom map if nothing else emerges in the meantime.