bondus/KlipperToolboard #49: Steps required to restore a damaged bootloader

ssvenn, 2024-03-09:
I bought a used Huvud board that I accidentally bricked with a bad flash update, it had an old version of CanBoot on it from the previous owner and I compiled Klipper with the wrong bootloader offset (2k instead of 8k), and the CanBoot bootloader happily overwrote part of itself while flashing. 

I didn't find any clear documentation on how to restore the bootloader on the Huvud but have been looking at the datasheet for the STM32F103 and it seems this chip does not support bootloader recovery through USB, only UART. And those UART pins are in use by the stepper driver I believe, while the pins that go to the jumper header block are used for the stock bootloader?

Am I right in thinking that the chip would have to be desoldered to be reflashed?

bondus, 2024-03-09:
You can use a ST-Link programmer to flash a new bootloader. The 6-pin header has the necessary pins. You can find cheap clones "ST-Link v2" on most 
You might have to connect the test points boot0 "B0" and 3.3V on the backside to force the CPU into the ROM bootloader.

ssvenn, 2024-04-02:
Thanks, can confirm that a clone ST-Link programmer was able to erase and reflash the board!

closed 2024-04-02 by ssvenn